first95/FRC2017 #21: Feature: plan for autonomous moves

lindsayvallen, 2017-02-14:
Here's a document describing four main autonomous moves:
https://docs.google.com/document/d/1ENUvHZCirlT5yJ7iKj6ez_yQ3wbROgJhQJvOVjOUkFM/edit?usp=sharing

Input welcome from all. @first95/developers @first95/admins

jwalthour, 2017-02-19:
This is unrelated but has some additional diagrams of approach paths; for help in discussions: https://docs.google.com/document/d/1LGmk7nhMmKP2ZSOrxvcwZcQ96EXlEnTz25DGXv2CLmg/edit?usp=sharing

jwalthour, 2017-02-22:
When approaching from a position snugged up against the boiler with the back of the robot square against the back wall, the sequence is:
1. Drive forward 88.44" - [half robot length]
2. If on the blue side of the field, turn 60 degrees clockwise.  If on the red side of the field, turn 60 degrees counter clockwise.
3. Drive forward 82.56" - [half robot length] - [camera standoff distance]
4. Activate ScoreGear visual-based sequence.

When approaching from a position snugged up against the loading station with the back of the robot square against the back wall, the sequence is:

1. Drive forward 81.177" - [half robot length]
2. If on the blue side of the field, turn 60 degrees clockwise.  If on the red side of the field, turn 60 degrees counter clockwise.
3. Drive forward 84.84" - [half robot length] - [camera standoff distance]
4. Activate ScoreGear visual-based sequence.

jwalthour, 2017-02-22:
I believe planning to be complete.

closed 2017-02-22 by jwalthour