first95/FRC2017 #39: Dead reckoning does not travel predictable distances

jwalthour, 2017-04-03:
Seems to have ceased to work properly sometime between the MABOS event and the SNH event.  Affects both driving straight and turning.  Have checked the following:

- Encoders appear properly mounted
- Encoders appear connected
- The new gear ratio doesn't affect it, because the encoders are on the gearbox output shaft
- The automove is reporting that it's reached its destination

jwalthour, 2017-04-03:
Here's an example of a different strategy, where PID positional control is delegated to the Talons: https://github.com/AquidneckIslandRobotics/2017/commit/3b4df11269cfa4c19e7b471545662c40c6b8efc9 

I'm not sure if Follower mode works properly with this design - I believe the code shown above is run on a drivetrain without ganged motors.

RuggS, 2017-04-13:
We have tested auto moves on Wilbur and all appears to be well. Enc ticks per foot has been changed and angle multiplier has been removed. We would like to test on Orville on carpet to confirm no differences between the 2 bots. Additionally we tested Boston code on Wilbur and it worked. Verifying this on Orville would also be good.